marcmerlin/AnimatedGIFs #2: Animation stalls when looping gif with >16 colors

rorosaurus, 2019-06-29:
Hello, thanks for this incredible library!  It's saved me tons of work and is very impressive!  
My setup is an ESP32 directly wired to a 128x64 1/32 HUB75E panel.  I quickly realized after diving into these libraries that feeding this panel is going to be a challenge.

Fortunately with my early tests, supplying a 128x64px gif with only 16 colors appears to be fairly smooth, with only a few frames stuttering per minute.  I suspected this wouldn't remain the case if I added more colors to the gif, and that seems to be the case.  Here is a test with 64 colors:
[image]

Curiously, the animation plays smoothly, it's only at the loop (frame 12 -> frame 1) that it stalls.  This leads me to think it could be related to allocating memory (which based on the forum posts I read seems like the big issue with these larger panels).  Do you happen to know if we re-allocate that memory when restarting the gif, and is that necessary?

rorosaurus, 2019-07-01:
Update: 32 colors is still produced smoothly.

marcmerlin, 2019-07-01:
Hi @rorosaurus , to be honest, I never tried gifs with fewer colors, that said I know the gif library was originally written for 32x32 resolution, with a way to go to 64x64. Past that, it's very possible that you're pushing some limits.
For memory, it is actually allocated once the first time a gif gets displayed and then the buffers stay around (if you look on the serial port, you'll see that), so it's not a malloc issue, but there are a few smaller arrays that get re-allocated within the gif library each time.
Then again, they are small enough that I'm not sure they'll cause the hang.
I'm afraid all I can say is trial and error, and if you get really stuck, you could try teensy instead which allocates memory differently and may work around that hang.
Louis Beaudoin may be able to help too, but then again my fork of his library has required malloc changes on ESP32, because his would likely not work at all with that much memory required.
Sorry I don't have more concrete help.

rorosaurus, 2019-07-01:
No worries, thank you for the info!  I will do some more digging and debugging and let you know what I find!

marcmerlin, 2020-11-27:
Hi @rorosaurus  this did get fixed in the end?
I now remember that SPIFFS also causes read hangs and that those went away when I switched to FATFS

marcmerlin, 2021-01-13:
this is an old bug, I'm pretty darn sure this is fixed by using FatFS instead of SPIFFS. Please let me know if it was another issue, or how you solved it

closed 2021-01-13 by marcmerlin