marcmerlin/AnimatedGIFs #5: Trying to get this to run with ws2812b strip and esp32

Davids365, 2020-02-22:
Hi, 

I'm currently trying to get this to run with an 12x12 ws2812b strip matrix, an esp32 with an SdCard Modul. Everytime I try to get this run I get this error message:

> In file included from C:\Users\FabE2\Documents\Arduino\libraries\SmartMatrix3\src/SmartMatrix3.h:175:0,
> 
>                  from sketch\animatedgif_config.h:38,
> 
>                  from sketch\FilenameFunctions.cpp:8:
> 
> C:\Users\FabE2\Documents\Arduino\libraries\SmartMatrix3\src/SmartMatrix_Impl.h:26:24: fatal error: DMAChannel.h: No such file or directory
> 
> compilation terminated.
> 
> exit status 1
> Error compiling for board ESP32 Dev Module.

I already tried to code out the Smartmatrix implementation, but then I start to get into deeper Code errors. Maybe I made some mistakes, which I didn't see, so I tried to add the missing DMAChannel.h and kinetis.h files, then it warns me that SmartMatrix3 isn't compatible with esp32:

> exit status 1
> no matching function for call to 'SmartMatrix3<24, 64, 96, 0u, 0u>::begin(int&)'

The error code is way to long, I can attach it if needed. I understand that the Smartmatrix3 library isn't compatible with the esp32, that's why I tried to code it out in the first place. So what did made wrong and what do I need to do to get the code working the esp32?

marcmerlin, 2020-02-23:
Did you edit 
https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h?
From the error, it clearly seems that you did not and you are using the default which is using SmartMatrix.
Please create your own config bit related to your array, you can use 
https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h#L576
as an example.

That said, I just want to be clear that 12x12 isn't going to let you display anything satisfying as far as gifs are concerned. you'd want 16x16 as a bare minimum and I'd recommend 32x32 if you can.
If after having properly configured neomatrix_config.h, you get other issues, please open a new issue with relevant details. I'll close this one.
Also, on ESP32, I recommend you use SPIFFS or FATFS instead of sdcard, unless 3MB of storage isn't big enough.

Davids365, 2020-02-23:
Thank you for answering.
I think I didn't made my Issue clear, I just started to learn programming.
The Issue is that when I define which Matrix I use (I also chosed the M64BY64), there's an error while compiling for the ESP32 Board:
[Error.TXT](https://github.com/marcmerlin/AnimatedGIFs/files/4241344/Error.TXT)

So I guess that I missed to edit something in the config or made a mistake that i don't understand. I tried the config with the SimpleGifAnimViewer sketch. There I get the same output, it doesn't fail to compile but nothing is shown on the matrix. 

The 12x12 is just the test at home for the bigger 64X64 project, and I need the Sd card so that I can just swap the card to change the animation, depending on the occasion.

marcmerlin, 2020-02-23:
Your error still shows that you are compiling for SmartMatrix.
You did not show me the neomatrix_config.h file you are using, so you're making me guess what you might have or not have done. 
My guess is that you didn't 
#define M64BY64
so https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h#L54
is true and then you get
https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h#L72
 #define SMARTMATRIX

MeisterLukas, 2020-02-24:
Hey I have a similar problem, here is my file: 

[neomatrix_config.TXT](https://github.com/marcmerlin/AnimatedGIFs/files/4247004/neomatrix_config.TXT)

Can you please look into it?

Ive got the #error "Please write a matrix config or choose one of the definitions above"

marcmerlin, 2020-02-24:
I was looking into it, I was seeing problems on ESP8266, but not ESP32 or arduino make.
It wasn't quite obvious what broke, I'm going to look into it some more.
@MeisterLukas  which CPU were you compiling for?

MeisterLukas, 2020-02-24:
Ok thank you,  I use an ESP32 from Espressif I think its the v.1

I  have it like this (the // got lost while converting to a .txt file):
    

>#ifdef ESP32
>     //#define ST7735_128b128
>     //#define ST7735_128b160
>     //#define ILI9341
>     //#define SMARTMATRIX
>     #define M64BY64
>     #endif
> #endif

But I get the same error even if I completly delete everything except M64BY64.

marcmerlin, 2020-02-24:
@MeisterLukas  ok, it's a bit hard to debug remotely without seeing what you're doing at each step.
Let me explain how this is supposed to work, which may be easier.

You are hitting this
https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h#L617
now scroll back up in the file, and look for these
https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h#L587
https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h#L557
etc...

because all those "#elif defined(M64BY64)" statements failed, you ended up on the last #else
If your file does have '#define M64BY64' on its own, it should be impossible to hit the else at the end.
Can you make really sure you've defined one of these?

If you have, please try 'git diff neomatrix_config.h' and paste the output to show me how your file is different from the stock one, or if you can't please post the whole file to https://gist.github.com/

Thanks

MeisterLukas, 2020-02-25:
I'm currently not at home and don't have the data with me, but I will check this when I am.

So that I understand you right everything between
https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h#L95
and
https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h#L587
is useless in my case and can be deleted, right?

But wouldn't the error just simply jump down to: https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h#L919 
if it's a not working board or a define error?
But I guess that I don't need the parallel output code anyway. Or do I, to define the Pinout?

marcmerlin, 2020-02-25:
@MeisterLukas no, that would be deleting too much and generally you shouldn't need to delete any lines, you simply have to define the right thing relevant to your array.
I'm trying to see if this file is too difficult to read/understand.
Are you generally familiar with how C/C++ define and ifdef works?
Can you see how all the blocks ifdef, elif, elif, elif are part of the same thing?
You simply need to define M16BY16T4 or M64BY64 at the top of the file and you don't need to delete anything.

Before you try anything else, since I assume you are using  FastLED_NeoMatrix , you should try the default demo:
https://github.com/marcmerlin/FastLED_NeoMatrix/tree/master/examples/MatrixGFXDemo
If you don't define anything, you'll end up with 
https://github.com/marcmerlin/FastLED_NeoMatrix/blob/master/examples/MatrixGFXDemo/MatrixGFXDemo.ino#L96
which is a 16x16 zigzag definition. You can modify it for your array and get that working first.

Back to your question, you probably want to use M16BY16T4 which does not use parallel output.
All you'd have to do is change MATRIXPIN and the neomatrix definition depending on whether you use zigzag, and what kind of tiling you use.
You can use M64BY64 for its lack of tiles (it's a straight wiring) and M16BY16T4 for its wiring on a single pin:
https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h#L887

marcmerlin, 2020-02-26:
@MeisterLukas @Davids365  please report if you can get 
https://github.com/marcmerlin/FastLED_NeoMatrix/blob/master/examples/MatrixGFXDemo/MatrixGFXDemo.ino#L96
working for you as it's the basic example to get neomatrix working

In the meantime, i'm reworking neomatrix_config for you to make it simpler to use/understand, as it's clear that it's not very simple

marcmerlin, 2020-02-27:
@MeisterLukas @Davids365 I have updated the code for you
Please git pull (update), and then edit this line
https://github.com/marcmerlin/AnimatedGIFs/blob/5107225160f1b88f9b1c66496ecd0e607fbe327a/neomatrix_config.h#L47
uncomment the M24BY24
and then edit the 2 places in the code where it's used and set the correct data for your matrix.
Hopefully this should be simpler for you to use.
Please an update if you have issues.

closed 2020-02-27 by marcmerlin

Davids365, 2020-02-27:
Hi, sorry for the late answer. 
For me the MatrixGFXDemo workes fine, I also used it to test my wiring before I used this code. 
I figuered out what my mistake was, I simply forgot to uncomment the Smartmatrix and deleted something mendatory in the code by accident. Your updated version works fine for me and is much easier to understand as an beginner. Thank you very much and sorry for the hussle I caused.

marcmerlin, 2020-02-27:
@Davids365 great to hear the new version is easier to use/understand. Sorry for the time you spent on the old one that wasn't as obvious (even if it was otherwise identical).

marcmerlin, 2020-03-04:
@MeisterLukas 
1) 128x128 is 16384 pixels. that would be very expensive and require about 400 amps of power if you use neopixels.
It is possible though, you'd just have to change a couple of numbers in the library

2) SDcard + spiffs, it is possible but you'd have to modify the code a fair bit as right now it's using ifdefs and not designed to use both at the same time (as in both cannot be compiled at the same time right now)