marcmerlin/AnimatedGIFs #9: Fat FS mount failed. Not enough RAM?

yannickh2, 2021-01-13:
Hello,

after I added the Gif files on the SPIFF and uploaded everything my Matrix is shining bright but displaying no animated GIFS.
 I updated neomatrix config.h and animatedgif_config.h
I get this output in the serial monitor.
` >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>> Serial.begin
18:22:35.343 -> Memory after setup() starts
18:22:35.343 -> Heap/32-bit Memory Available     : 355660 bytes total, 123248 bytes largest free block
18:22:35.343 -> 8-bit/malloc/DMA Memory Available: 287564 bytes total, 123248 bytes largest free block
18:22:35.343 -> After matrixleds malloc
18:22:35.343 -> Heap/32-bit Memory Available     : 359100 bytes total, 123248 bytes largest free block
18:22:35.378 -> 8-bit/malloc/DMA Memory Available: 286012 bytes total, 123248 bytes largest free block
18:22:35.378 -> Neomatrix total LEDs: 512 running on pin: 15
18:22:35.378 -> Before matrix->begin
18:22:35.378 -> Heap/32-bit Memory Available     : 356700 bytes total, 123248 bytes largest free block
18:22:35.378 -> 8-bit/malloc/DMA Memory Available: 283612 bytes total, 123248 bytes largest free block
18:22:35.378 -> Framebuffer_GFX::begin Width: 16 Height: 16 Num Pixels: 256
18:22:35.411 -> Setting Brightness: 64
18:22:35.411 -> Gamma Correction: 1.00
18:22:35.411 -> matrix_setup done
18:22:35.484 -> Matrix->show() Speed Test fps: 64
18:22:35.484 -> Fat FS mount failed. Not enough RAM?
`
I could not find anything on the error "Fat FS mount failed. Not enough RAM?" Do you know whats going wrong?
Thank you.

marcmerlin, 2021-01-13:
Hi.
This error happens in 
```
    #ifdef FSOFAT
        // Limit Fat support to a single concurrent file to save RAM
	// 37248 KB are saved by using limiting to 1 file instead of 10
        #ifdef GIFANIM_INCLUDE
	    // 1 file for reading files, and one file for the web server
	    if (!FFat.begin(0, "", 2)) die("Fat FS mount failed. Not enough RAM?");
        #else
	    // Need 2 file descriptors instead of 1 for file browser
	    if (!FFat.begin(0, "", 2)) die("Fat FS mount failed. Not enough RAM?");
        #endif
	Serial.println("FatFS Directory listing:");
    #else
```

Look in 
neomatrix_config.h:    #define FSOFAT
around that line, comment this out, and uncomment SPIFFS.

As a warning, SPIFFS will get slow and hang for short times on you if you have more than 1 or 2MB of data in it.

I'll update the documentation to make this more clear, thanks for the feedback.

closed 2021-01-13 by marcmerlin

marcmerlin, 2021-01-13:
https://github.com/marcmerlin/AnimatedGIFs/blob/master/README.md is now more up to date. Sorry that it didn't keep up with the code.
I only have limited time to update to many docs and pages, so if you find anything that can be improved, pull requests (even for documentation) are greatly appreciated :)

yannickh2, 2021-01-13:
Hi,
thanks for your reply and help.. I think update of the Readme helps, but I am still having a problem.
I commented out `//#define FSO FFat
    //#define FSOFAT` in neomatrix_config.h:
Now I am just getting this: "Error opening GIF File".
Serial Monitor:
`22:25:11.664 -> >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>> Serial.begin
22:25:11.698 -> Memory after setup() starts
22:25:11.698 -> Heap/32-bit Memory Available     : 355756 bytes total, 123344 bytes largest free block
22:25:11.698 -> 8-bit/malloc/DMA Memory Available: 287660 bytes total, 123344 bytes largest free block
22:25:11.698 -> After matrixleds malloc
22:25:11.698 -> Heap/32-bit Memory Available     : 359196 bytes total, 123344 bytes largest free block
22:25:11.698 -> 8-bit/malloc/DMA Memory Available: 286108 bytes total, 123344 bytes largest free block
22:25:11.735 -> Neomatrix total LEDs: 512 running on pin: 15
22:25:11.735 -> Before matrix->begin
22:25:11.735 -> Heap/32-bit Memory Available     : 356796 bytes total, 123344 bytes largest free block
22:25:11.735 -> 8-bit/malloc/DMA Memory Available: 283708 bytes total, 123344 bytes largest free block
22:25:11.735 -> Framebuffer_GFX::begin Width: 16 Height: 16 Num Pixels: 256
22:25:11.735 -> Setting Brightness: 64
22:25:11.735 -> Gamma Correction: 1.00
22:25:11.735 -> matrix_setup done
22:25:11.835 -> Matrix->show() Speed Test fps: 64
22:25:11.902 -> Directory listing:
22:25:12.004 -> FS File: /32anim_dance.gif Size: 62866
22:25:12.040 -> 
22:25:12.040 -> GifAnim Viewer enabled, lzwMaxBits: 12, gif_size: 64
22:25:12.040 -> /32anim_dance.gif: Error opening GIF file
`

I checked and it is coming fromGifAnim_impl.h 
`#else
    if (file) file.close();
    #ifdef FSOSPIFFS
	file = SPIFFS.open(pathname, "r");
    #else
	file = FFat.open(pathname);
    #endif
#endif
    if (!file) {
        Serial.println(": Error opening GIF file");`

I changed the pathname in the last else in SimpleGifAnimViewer2
`#else
	const char *pathname = "/32anim_dance.gif";
    #endif`

In my sketch data folder on the SPIFF on my ESP32 is the 32anim_dance.gif

Thanks for your help.

marcmerlin, 2021-01-13:
Did you uncomment SPIFFS in the same file you commented out FatFS?

Which demo are you using?

Try AnimatedGifs.ino, it does a directory listing for you first:
```
Heap/32-bit Memory Available     : 204604 bytes total, 64624 bytes largest free block
8-bit/malloc/DMA Memory Available: 142108 bytes total, 64624 bytes largest free block
Total PSRAM used: 0 bytes total, 4194252 PSRAM bytes free
Framebuffer_GFX::begin Width: 64 Height: 96 Num Pixels: 6144
Setting Brightness: 255
Gamma Correction: 1.00
matrix_setup done
Matrix->show() Speed Test fps: 52
FatFS Directory listing:
Total space:   12378112
Free space:     9232384
FS File: /gifs64/087_net.gif Size: 46200
FS File: /gifs64/149_minion1.gif Size: 48766
FS File: /gifs64/193_redplasma.gif Size: 25083
FS File: /gifs64/196_colorstar.gif Size: 79754
FS File: /gifs64/200_circlesmoke.gif Size: 48029
FS File: /gifs64/203_waterdrop.gif Size: 31736
FS File: /gifs64/210_circletriangle.gif Size: 46607
FS File: /gifs64/215_fallingcube.gif Size: 70478
```

marcmerlin, 2021-01-13:
Directory listing code is in GifAnim_Impl.h

If you have never used SPIFFS, I recommend you get very basic code working first to make sure that you list and see files in the right place.

yannickh2, 2021-01-13:
I uncommented SPIFF in the same file as I uncommented FatFS
in neomatrix_config.h 
`#elif defined(ESP32)
    #define FS_PREFIX ""
    #include <SPIFFS.h>
    #define FSO SPIFFS
    #include "FFat.h"
    //#define FSO FFat
    //#define FSOFAT
    // Do NOT add a trailing slash, or things will fail
    #if gif_size == 64
        #define GIF_DIRECTORY FS_PREFIX "/gifs64"
    #else
        #define GIF_DIRECTORY FS_PREFIX "/gifs"
    #endif`

I also get the directory reading if I add more gifs to the SPIFF:
` Directory listing:
23:14:43.406 -> FS File: /gifs/cubeconstruct.gif Size: 25745
23:14:43.406 -> FS File: /gifs/32anim_photon.gif Size: 21221
23:14:43.439 -> FS File: /gifs/32anim_dance.gif Size: 62866
23:14:43.439 -> FS File: /gifs/circles_swap.gif Size: 21864
23:14:43.439 -> FS File: /gifs/cubeslide.gif Size: 48466
23:14:43.439 -> FS File: /gifs/corkscrew.gif Size: 17529
23:14:43.439 -> FS File: /gifs/32anim_flower.gif Size: 6784
23:14:43.439 -> FS File: /gifs/concentric_circles.gif Size: 14582
23:14:43.473 -> 
23:14:43.473 -> GifAnim Viewer enabled, lzwMaxBits: 12, gif_size: 64
23:14:43.473 -> /gifs/32anim_dance.gif: Error opening GIF file
`

Using the demo SimpleGifAnimViewer2

marcmerlin, 2021-01-13:
So, doesn't that tell you what's wrong?

You use "/32anim_dance.gif"
and
you have "/gifs/32anim_dance.gif"

yannickh2, 2021-01-13:
No sorry, I have the in SimpleGifAnimViewer `const char *pathname = "/gifs/32anim_dance.gif";`
and in serial monitor too 
`FS File: /gifs/cubeconstruct.gif Size: 25745
23:24:37.468 -> FS File: /gifs/32anim_photon.gif Size: 21221
**23:24:37.468 -> FS File: /gifs/32anim_dance.gif Size: 62866**
23:24:37.468 -> FS File: /gifs/circles_swap.gif Size: 21864
23:24:37.468 -> FS File: /gifs/cubeslide.gif Size: 48466
23:24:37.468 -> FS File: /gifs/corkscrew.gif Size: 17529
23:24:37.502 -> FS File: /gifs/32anim_flower.gif Size: 6784
23:24:37.502 -> FS File: /gifs/concentric_circles.gif Size: 14582
`

marcmerlin, 2021-01-13:
Ok, you lost me, you've changed code again (animviewer2, now animviewer)

It's a bit hard to debug this remotely. can you try some local troubleshooting?
Can you write or get basic code that opens the gif and ouputs its first bytes on serial console?
Does it work? etc...