marcmerlin/AnimatedGIFs #10: Fat FS mount failed. Not enough RAM?

dallen98, 2021-02-08:
Hello Marc
I have Animatedgifs working on raspberry pi using the default 4MB with FFAT(1.2mb APP/1.5mb FAT) I am using Lorol/arduino-esp32fs-plugin to upload the gifs.

I am now trying to get No OTA 1mbAPP/3MB fats to work on raspberry pi and get the following output..

Memory after setup() starts
Heap/32-bit Memory Available     : 349432 bytes total, 122256 bytes largest free block
8-bit/malloc/DMA Memory Available: 286632 bytes total, 122256 bytes largest free block
Total PSRAM used: 0 bytes total, 4194252 PSRAM bytes free
After matrixleds malloc
Heap/32-bit Memory Available     : 349432 bytes total, 122256 bytes largest free block
8-bit/malloc/DMA Memory Available: 286632 bytes total, 122256 bytes largest free block
Total PSRAM used: 12288 bytes total, 4181948 PSRAM bytes free
Neomatrix 16 pin via RMT/I2S 16 way parallel output, total LEDs: 4096
Before matrix->begin
Heap/32-bit Memory Available     : 346372 bytes total, 122256 bytes largest free block
8-bit/malloc/DMA Memory Available: 283572 bytes total, 122256 bytes largest free block
Total PSRAM used: 12288 bytes total, 4181948 PSRAM bytes free
Framebuffer_GFX::begin Width: 64 Height: 64 Num Pixels: 4096
Setting Brightness: 128
Gamma Correction: 2.40
matrix_setup done
Matrix->show() Speed Test fps: 125
Fat FS mount failed. Not enough RAM?

I then edit noota_3gffat.csv as below with ffat,     data, fat,     0x111000,0x2F0000,
and #ffat,     data, fat,     0x111000,0x2EF000,

nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x100000,
ffat,     data, fat,     0x110000,0x2F0000,
# to create/use ffat, see https://github.com/marcmerlin/esp32_fatfsimage
#ffat,     data, fat,     0x111000,0x2F0000,
#ffat,     data, fat,     0x111000,0x2EF000,

and still get ...Fat FS mount failed. Not enough RAM?
Perhaps you can point me in the right direction thanks Dave

marcmerlin, 2021-02-09:
I'm going to assume you meant ESP32 and not rPI :)
Please have a look at the updated https://github.com/marcmerlin/esp32_fatfsimage
https://github.com/marcmerlin/esp32_fatfsimage/blob/master/arduino/ffat/ffat.ino
will give you an idea of where the partitions are in case a number is wrong

That said, I now recommend you use https://github.com/lorol/arduino-esp32fs-plugin this should simply create the FatFS image from the arduino IDE

dallen98, 2021-02-09:
Sorry I meant to say I have the https://github.com/lorol/arduino-esp32fs-plugin loaded on the rpi.

My objective is to run a single animated gif 15 seconds long on the esp32 so that in mind does the Fatfs offer any advantage over spiffs ???

Alternativley I could run 3 x  5second long animated gifs instead of the above if it makes life easier ?
Let me know your thoughts Thanks

marcmerlin, 2021-02-09:
You still said rpi instead of ESP32 :)
15 seconds on SPIFFS should work absolutely fine, so you could switch to that
Have you tried https://github.com/marcmerlin/esp32_fatfsimage/blob/master/arduino/ffat/ffat.ino to debug things?

dallen98, 2021-02-09:
I can upload spiffs no problem using https://github.com/lorol/arduino-esp32fs-plugin so sounds as though thats all I need then for my 15 second application, I would add that its somewhat big 360x180 pixels

[image]

dallen98, 2021-02-09:
...and I am trying to merge this into Yves Bazins Virtual driver

marcmerlin, 2021-02-09:
Ok, so let's recap:
1) fatfs should work, especially with  https://github.com/lorol/arduino-esp32fs-plugin  . I gave you some code to use for debugging. I do use it, so it works, but if you can't figure out what's wrong, I can try to help debug further
2) spiffs should work just as well for your application
3) if you are having issues with either, please try very simple code that opens the filesystem and lists the content
4) Yves is awesome, hopefully you get your code working with his stuff

dallen98, 2021-02-09:
so spiffs route it is then, thanks for your help

marcmerlin, 2021-02-09:
fatfs should work just fine given the plugin you are using, but you haven't given me more debugging information, so I can't tell you more :)
spiffs uses the same mechanisms, but is older so maybe it'll work better for you (especially if your ESP32 hardware layer checkout is older)

dallen98, 2021-02-09:
yes if the fatfs doesnt add anything for me then i,m reluctant to spend time on it, does the littlefs offer any read speed benefits over the spiffs for my particular scenario?

marcmerlin, 2021-02-09:
littlefs is supposed to be:
- smaller than fatfs in memory use
- better with flash wear
- without the performance issues of spiffs
I have not used it though, as it didn't exist when I wrote my code

dallen98, 2021-02-09:
Do you know any fastled experts that code for payment ?

marcmerlin, 2021-02-10:
you should ask here https://www.reddit.com/r/FastLED/

closed 2021-04-22 by dallen98