marcmerlin/AnimatedGIFs #11: Not an issue. Just a quick question. I have successfully managed to run 64x64 gif Animation example code. But when I try to run 128x64 gif on 4 chained  64x32 panels . Then it only playing animation in 64x64 area rest 64x64 panel area is blank. How do I could run 128x64 animation? There is no option I am able to locate to change gif width and height in neoconfig.h file.  Any help in this regard will be appreciated.

HarpreetHeera, 2021-03-06:


marcmerlin, 2021-03-06:
You're looking for gif_size: https://github.com/marcmerlin/AnimatedGIFs/blob/179a2d6593f073beb29cbaf9bffe06a3f91b95ab/neomatrix_config.h#L151

I think the library expects square gifs, so you'd need to decode a 128x128 gif and use the OFFSETY as shown in https://github.com/marcmerlin/AnimatedGIFs/blob/179a2d6593f073beb29cbaf9bffe06a3f91b95ab/SimpleGifAnimViewer/SimpleGifAnimViewer.ino#L7

Let me know if that works and I'll close this

HarpreetHeera, 2021-03-06:
Thanks Marc Merlin for very quick response. Please give me few days for this to work on your suggestion. I will update you with in next week. 

Many Thanks. :)

HarpreetHeera, 2021-03-09:
Hi MarcMerlin,
Thank you so much for your suggestions, but it was my dumb mistake. Rectangular gifs are also working smoothly. 
I was doing following mistake. 
In [https://github.com/marcmerlin/AnimatedGIFs/blob/master/neomatrix_config.h](url) file I was passing direct pixels count in 
`    const uint8_t kMatrixWidth = 128;
    const uint8_t kMatrixHeight = 64;`

But it works well when I passed Tile size first then total number of Tiles in horizontal and vertical.  Actually 
I was thinking , these configuration was for NeoMatrix not for SmartMatrix. But In reality these variables are also used for mapping smartMatrix kpanels.

`const uint16_t MATRIX_TILE_WIDTH = 64; // width of EACH NEOPIXEL MATRIX (not total display)
    const uint16_t MATRIX_TILE_HEIGHT= 32; // height of each matrix
   // Used by LEDMatrix
    const uint8_t MATRIX_TILE_H     = 2;  // number of matrices arranged horizontally
    const uint8_t MATRIX_TILE_V     = 2;  // number of matrices arranged vertically
    
    // Used by NeoMatrix
    const uint16_t mw = MATRIX_TILE_WIDTH *  MATRIX_TILE_H;
    const uint16_t mh = MATRIX_TILE_HEIGHT * MATRIX_TILE_V;
    
    /// SmartMatrix Defines
    #define COLOR_DEPTH 24         // known working: 24, 48 - If the sketch uses type `rgb24` directly, COLOR_DEPTH must be 24
    const uint8_t kMatrixWidth = mw; //128
    const uint8_t kMatrixHeight = mh; //64`
Many Thanks again for your quick support and nice work.
Regards,
Harpreet

closed 2021-03-09 by HarpreetHeera

marcmerlin, 2021-03-09:
glad you got it working. To be fair, there are 2 different codebases
1) neomatrix
2) the gif code
so indeed they need to be configured separately and correctly for everything to work :)