marcmerlin/AnimatedGIFs #12: 60 x 120px Doesnt Display on M64BY64

dallen98, 2021-04-22:
Re M64BY64 I,m using a ws2812b matrix 60 leds wide x 15 leds high and cannot get anymore than 64 leds high to show up, I see a similar issue being discussed here although using rgb panels...
https://github.com/marcmerlin/AnimatedGIFs/issues/11
here is my setup..

#elif defined(M64BY64) // 64x64 straight connection (no matrices)
    #include <FastLED_NeoMatrix.h>
    #define FASTLED_NEOMATRIX
    
    const uint16_t MATRIX_TILE_WIDTH = 60; // width of EACH NEOPIXEL MATRIX (not total display)
    const uint16_t MATRIX_TILE_HEIGHT= 120; // height of each matrix
    const uint8_t MATRIX_TILE_H     = 1;  // number of matrices arranged horizontally
    const uint8_t MATRIX_TILE_V     = 1;  // number of matrices arranged vertically
    #define NUM_STRIPS 5                    //go to line 763// NUM_STRIPS pins////////////////////////////////////////////////////////////
    #define NUM_LEDS_PER_STRIP 900  //rio, no more than 900 leds per pin or it fails
    //#define NUM_LEDS_PER_STRIP 256
    // Used by NeoMatrix
    const uint16_t mw = MATRIX_TILE_WIDTH *  MATRIX_TILE_H;
    const uint16_t mh = MATRIX_TILE_HEIGHT * MATRIX_TILE_V;
    
    CRGB *matrixleds;
    #ifdef LEDMATRIX
    // cLEDMatrix defines
    cLEDMatrix<MATRIX_TILE_WIDTH, MATRIX_TILE_HEIGHT, VERTICAL_ZIGZAG_MATRIX> ledmatrix(false);
    #endif
    FastLED_NeoMatrix *matrix = new FastLED_NeoMatrix(matrixleds, MATRIX_TILE_WIDTH, MATRIX_TILE_HEIGHT,
        NEO_MATRIX_TOP + NEO_MATRIX_LEFT +    //rio
        NEO_MATRIX_ROWS + NEO_MATRIX_ZIGZAG);
        //NEO_MATRIX_COLUMNS + NEO_MATRIX_ZIGZAG );
    
//=======================================================================
    #elif defined(M64BY64) // 64x64 straight connection (no matrices)
        // https://github.com/FastLED/FastLED/wiki/Multiple-Controller-Examples
        FastLED.addLeds<WS2812B,23, GRB>(matrixleds, 0*NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP);
        FastLED.addLeds<WS2812B,22, GRB>(matrixleds, 1*NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP);
        FastLED.addLeds<WS2812B,27, GRB>(matrixleds, 2*NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP);
        FastLED.addLeds<WS2812B,21, GRB>(matrixleds, 3*NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP);
        FastLED.addLeds<WS2812B,19, GRB>(matrixleds, 4*NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP);
//        FastLED.addLeds<WS2812B,18, GRB>(matrixleds, 5*NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP);

So just to be clear pins 23,22,27,21, correctly display figures 1,2,3,4 in the 60w x 120h gif below but pin 19 displays only the top 4 pixel lines on the figure 5, perhaps I need to alter the code somewhere other than neomatrix_config.h  ??

[image]

any help much appreciated

marcmerlin, 2021-04-25:
Ok, before we go anywhere, can you make sure that the neomatrix part is working correctly?
Can you run this demo and make sure it's working correctly?
https://github.com/marcmerlin/FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos/tree/a98cb3e965433a8eebb8893d630735fda12469cb/FastLED/PlasmaAnimation

you can also try code from https://github.com/marcmerlin/FastLED_NeoMatrix/blob/master/examples/MatrixGFXDemo/MatrixGFXDemo.ino
or copy this function and run it: https://github.com/marcmerlin/FastLED_NeoMatrix/blob/40fb822377497a7be8e1103be344104870ff0878/examples/MatrixGFXDemo/MatrixGFXDemo.ino#L363

Next if this is working, the gif decoder only handles square gifs, but you can have a square gif that isn't filled in and use OFFSETX/OFFSETY to place it where you want on your matrix

dallen98, 2021-04-28:
I have got the https://github.com/marcmerlin/FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos/tree/a98cb3e965433a8eebb8893d630735fda12469cb/FastLED/PlasmaAnimation sketch working, it shows mesmerising patterns on 6 pins...
    const uint16_t MATRIX_TILE_WIDTH = 60; // width of EACH NEOPIXEL MATRIX (not total display)
    const uint16_t MATRIX_TILE_HEIGHT= 120; // height of each matrix
    const uint8_t MATRIX_TILE_H     = 1;  // number of matrices arranged horizontally
    const uint8_t MATRIX_TILE_V     = 1;  // number of matrices arranged vertically
    #define NUM_STRIPS 6                    
    #define NUM_LEDS_PER_STRIP 900

Regarding MatrixGFX I am using the MatrixGFXDemo64 where it reaches count_pixels(); it displays alternating
red, green, blue strips indefinatly so i commented it out //count_pixels(); and then the rest of the functions smileys, scrolling text etc all seem to work ok.

I have below a 120x120px test gif which is divided into 16 coloured tiles, each tile is 60px wide x 15px high which I hope to display gifs on, thanks for your help.
[image]

marcmerlin, 2021-05-02:
Ok, so just to be clear,you're saying that count_pixels() works perfectly in the right order, that's good news, it means that your wiring and hardware are good.
So your problem is just an issue with the gif display. Again, the gif code is only tested with square gifs, so make sure you feed it a square gif and use OFFSETX/OFFSETY if needed, and it should display in the right location.
Does that work, or not?
If not, please post your code and a picture of what's not working right (although please do your own debugging and testing before you do, I only have very limited time to help others with their code, and by that I mean virtually none :) ).

marcmerlin, 2021-05-02:
Just to be clear, you can display a 120x120 gif (if it fits in RAM), and give it a negative offset OFFSETX=-30 so that it displays on a 60x120 array. The code is fine with displaying pixels out of the addressable zone, so your gif can be bigger than your display and you can pan it around to display the section you'd like.

dallen98, 2021-05-05:
Which tab and line number do I place OFFSETX=-30  ??
Many Thanks

marcmerlin, 2021-05-05:
please look at example code 
https://github.com/marcmerlin/AnimatedGIFs/blob/master/SimpleGifAnimViewer/SimpleGifAnimViewer.ino

dallen98, 2021-05-05:
Do you mean line 18...
// 32x32 GIFs on 24x32 display, hence offset of -4
    OFFSETX = -4;
in SimpleGifAnimViewer.ino ? Thanks

marcmerlin, 2021-05-05:
yes. Please don't take this the wrong way, but I can't do 1-1 support (well, I can, but I'm not doing the paid work I'm supposed to do instead), it's up to you also read the code and try things on your own.

dallen98, 2021-05-05:
ok thanks

closed 2021-05-05 by marcmerlin