marcmerlin/AnimatedGIFs #13: SDcard

pepster01, 2021-07-04:
Hi Marc,
I want to run the ESP8266 with the SD card so I can play gifs, how to setup and what are the pins to connect to the esp8266.
I know you can use the onboard memory but still not enough, also if I try to upload with (SPIFFS) I get SPIFFS Create Failed!
I would preferer to run them on SD card.
help would be appreciate.
Thanks

marcmerlin, 2021-07-04:
On an ESP8266 with 4MB of RAM, you can have 3MB of storage for SPIFFS, which has been enough for me. Is there a chance you have a 1MB ESP8266 and only 384KB for SPIFFS maybe?

You should be able to get an sdcard working, but you'll be on your own. The code is there, but was only tested on teensy and on ESP32 by someone else than me. If you can get a basic ESP8266 sdcard example (i.e. not my code) working, it should be easy enough to adapt it to my code.
That being said, you could also just get an ESP8266 that allows you to have 3MB of SPIFFS, that's a lot of storage for an animated gifs in size 32x32 or even 64x64

pepster01, 2021-07-04:
Ok, Thanks for that.

Superassi, 2022-06-07:
> Hi Marc, I want to run the ESP8266 with the SD card so I can play gifs, how to setup and what are the pins to connect to the esp8266. I know you can use the onboard memory but still not enough, also [...]

Did you get it?

I am completly confused whereI should define the correct Pins. SD_CS is okay, but I don't find any initialisation of the SD card like SD.begin, where i can change the Pins

marcmerlin, 2022-06-07:
@Superassi as per my prior answer, I have never used the sdcard code since I don't need it, but it should work with maybe a simple fix or two?  The begin is there
```
FilenameFunctions_Impl.h:    if (!SD.begin(chipSelectPin)) die("SD Begin failed");
```

That said, I recommend you look for a simple tutorial of sdcard on ESP8266 first, get that working, and once it works, try my code again. If it needs a simple patch, send it over and I'll merge it.

davemaster, 2022-06-07:
there are MANY examples how to interface sdcard to esp32... just google a little. I find what library specifically I did use to sucess: I call gif files from sdcard, then I show them in an RGB 32x32 panel