marcmerlin/AnimatedGIFs #14: "'FFat' was not declared in this scope" in SimpleGifAnimViewer

retrohedron, 2021-12-07:
Hey Marc,

First, thanks for all your work on this, it's super cool what kind of projects it's enabled.

Second, sorry if this is a dumb/simple issue, I'm pretty new to this side of programming.

So my intention is to hookup a 4 x 4 array of 8x8 WS2812B led matrices, but for now I'm just working with one to test all the hardware/software. I'm working on an Arduino Mega R3 (maybe a different board would be better?).

I uncommented the #define M24BY24 line in neomatrix_config.h, seemed like the closest match for now just to get something running.

Here's the error I'm getting on compile:

```
In file included from C:\Users\JJ\Documents\Arduino\AnimatedGIFs-master\SimpleGifAnimViewer\neomatrix_config.h:130:0,
                 from C:\Users\JJ\Documents\Arduino\AnimatedGIFs-master\SimpleGifAnimViewer\SimpleGifAnimViewer.ino:2:
C:\Users\JJ\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\JJ\Documents\Arduino\AnimatedGIFs-master\SimpleGifAnimViewer\SimpleGifAnimViewer.ino: In function 'void setup()':
SimpleGifAnimViewer:89:9: error: 'FFat' was not declared in this scope
  file = FFat.open(pathname);
         ^~~~
exit status 1
'FFat' was not declared in this scope
```

Thanks for any help/advice you can provide.

marcmerlin, 2021-12-07:
This is failing because animatedgifs wants to read the GIF from some storage, either SPIFFS or FFAT on ESP8266/ESP32 using internal flash, or an sdcard.
By default it will use FFAT (there are defines to change that). There is no support for this on arduino uno or mega, which is why it's not working.
So yes, a different board will be better. You can get an ESP32 for $5 to $10, much cheaper than your arduino mega, and it'll come with 4 to 16MB of flash storage

closed 2021-12-07 by marcmerlin

marcmerlin, 2021-12-07:
I should add that in theory, you could probably add an sdcard to a mega, but you'd probably have to modify and hack the code a bit to make that work. I wouldn't be able to help you with making that work

retrohedron, 2021-12-08:
Ahhhh, I'm glad my suspicions were at least pointed in the right direction. I'll try the ESP32 and see what happens. Thanks!

marcmerlin, 2021-12-08:
If you have never used an ESP32 before, I recommend you run basic example code to get FFAT working first (or SPIFFS, or the newer LittleFS that can be added to my code with minimal work).
SPIFFS will be a bit easier to setup, I recommend you just get a basic example working (sending a file to flash, and reading it back, you can google for it)