marcmerlin/AnimatedGIFs #15: How to build the project.

designdevy, 2022-01-20:
Dear Marc,

Hope this message finds you well. I am not sure how to build the SimpleGifAnimViewer example for ESP32. It says
[image]

Is it possible for you to provide the relevant libraries in one folder instead of links. I have been trying to find libraries that are referenced many times that might be installed on your system but not on mine so my project doesn't build.

Any help is appreciated. I hope I can animate this thing (also confused whether I should choose M16BY16T4 or ESP32 or both as #define:
[image]

marcmerlin, 2022-01-20:
Hi, are you on windows by any chance?

marcmerlin, 2022-01-20:
For the define, have a look at
```
// vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
// README README README README README README README README README README  README README README README
// If you have never used FastLED::NeoMatrix before, please try these 2 examples first
// https://github.com/marcmerlin/FastLED_NeoMatrix/tree/master/examples/matrixtest
// https://github.com/marcmerlin/FastLED_NeoMatrix/tree/master/examples/MatrixGFXDemo
// For FastLED just uncomment this line below (define M24BY24) and fix the matrix definition
// or use one of the other ones if they are closer ot your setup (M32BY8X3 M16BY16T4 M64BY64.
// For SmartMatrix, just use "#define SMARTMATRIX"
```

I'm not sure how your pixels are layed out. If they are in a straight panel, you'll probably want M24BY24 and fix it to be 32x32.
If they are 4 matrices of 16x16 put to together as a 32x32, you'l probably want  M16BY16T4

designdevy, 2022-01-21:
Hi Marc,

I am so glad you were able to reply. I was reluctant to approach since you are quite busy with stuff. Anyway, I am able to make some of your animations work from this repo:
https://github.com/marcmerlin/FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos

Pacman still doesn't work as it has a missing library.

I uncommented the M24BY24 and changed the settings to 16, 16, 2, 2 and others for the layout TOP + LEFT etc that had worked in simpletest and GFXTest as you had suggested because M16BY16T4 gave me some build errors.

I am trying to build it on a RaspberryPi and it doesn't work. I bet that is a linux but not sure why.

designdevy, 2022-01-21:
I will try to do the same today with GIFAnimations as that is what i built this display for. I am still trying to figure out how to upload the gif as a file system to an esp32 as I am new to filesystems in general but I have followed this project for quite a while and it was ported to PlatformIO for VS Code and uploading FS is a breeze. You might wanna check it out:
https://github.com/CODeRUS/GyverLampCpp

marcmerlin, 2022-01-21:
I am very busy, I can't help people with their code, but I do my best to help people with the code I published, otherwise it's not very useful if you can't use it ;)
Pacman needs the ledsprite library from github, but you can skip it. Glad you got the rest working by removing the symlinks.
"gave me build errors" and "doesn't work" is not very precise :) but hopefully you can figure it out.

For the filesystem, try SPIFFS and the arduino plugin: https://github.com/lorol/arduino-esp32fs-plugin/releases